open-ephys/bonsai-onix1 #490: Create StimulusSequence as an Observable

bparks13, 2025-08-25:
First referenced in #459, this is regarding the possibility of creating a separate `StimulusSequence` class that contains all parameters of a stimulus sequence, such as for a `Headstage64ElectricalStimulator`.

The main core of this issue is how to subscribe to changes in nested properties if we were to create a wrapper class for all sequence parameters. Currently, the original behavior of the `Headstage64ElectricalStimulatorTrigger` node, as well as the proposed behavior after adding the GUIs, is that each parameter is an individual property and hooks into an internal `BehaviorSubject` whenever it's `set` method is called; this allows the `Process()` method to subscribe to changes in the parameter and write to the appropriate register during acquisition.

When a wrapper class is introduced, the `set` method is only called if the entire sequence is being modified, such as by externalizing the property and updating the entire sequence in one go. If a nested property is modified, only the property-specific `set` method is called (e.g., `StimulusSequence.PowerEnable`), not the sequence `set` method. 

We will not give the `StimulusSequence` a `Device` or a `Context`, as it has no need for that information. Therefore, we need a way for the class to register any changes to individual parameters, which the device class can then subscribe to and determine which parameter has been modified to then write to the appropriate register(s).

## Proposal (Snapshot)

Researching the patterns used a little bit more, we might be able to implement it in the following manner:

1. Create an internal `BehaviorSubject` in the `StimulusSequence` class, and initialize it to the default parameters of the class.

```C#
public class Headstage64ElectricalStimulatorSequence : IObservable<StimulusSequence>
{
    readonly BehaviorSubject<Headstage64ElectricalStimulatorSequence> sequence;

    public Headstage64ElectricalStimulatorSequence()
    {
        sequence = new(this);
    }
}
```

2. Whenever an individual property changes, send an update via `OnNext` using a snapshot of the entire sequence:

```C#
uint triggerDelay;
public uint TriggerDelay
{
    get => triggerDelay;
    set
    {
          triggerDelay = value;
          sequence.OnNext(this);
    }
}
```

3. Subscribe to these changes, and expose a method for the device to subscribe to these changes as well, following the established pattern in the `ObservableExtensions.SubscribeSafe()` extension:

```C#
public IDisposable SubscribeSafe(IObserver<Headstage64ElectricalStimulatorSequence> observer, Action<Headstage64ElectricalStimulatorSequence> action)
{
    return sequence.SubscribeSafe(observer, action);
}
```

4. At the level of the device, this would look like:

```C#
public Headstage64ElectricalStimulatorSequence StimulusSequence { get; set; }

// [...]

public override IObservable<ContextTask> Process(IObservable<ContextTask> source)
{
    var enable = Enable;
    var deviceName = DeviceName;
    var deviceAddress = DeviceAddress;
    return source.ConfigureDevice((context, observer) =>
    {
        var device = context.GetDeviceContext(deviceAddress, DeviceType);
        device.WriteRegister(Headstage64ElectricalStimulator.ENABLE, enable ? 0u : 1u);

        return new CompositeDisposable(
            StimulusSequence.SubscribeSafe(observer, value => /* Parse the sequence here, only write updated parameters */),
            DeviceManager.RegisterDevice(deviceName, device, DeviceType));
    });
}
```

## Proposal (Diff)

If we want more granular control over the subscription process, and to minimize the need to process the entire sequence to figure out which elements were updated, it seems like we could extend the Snapshot proposal and create a `BehaviorSubject` for each parameter, write to the `OnNext` for that particular subject whenever an individual parameter is updated, and expose a subscription method for that parameter. However, this method would introduce a lot of boiler plate, and might not be of huge benefit to us. I'm leaving it up as a point for discussion in case it ends up being beneficial.

## Final Thoughts

This may not end up being the best solution, as I'm not sure how well it will work in the larger context of the devices/headstages, but it seems like this method should work for any property changes that occur inside of the Sequence class. There may be other scenarios that I am not accounting for, so please let me know if there are any additional considerations we should keep in mind while developing this.

One thing I am not confident about the proposal is how it will handle the entire sequence being updated, if it is externalized; I don't think this will work, but I have not spent the time researching how to implement that yet.

aacuevas, 2025-08-26:
After giving it some thought, I believe the best, most idiomatic and generalizable option is [`INotifyPropertyChanged`](https://learn.microsoft.com/en-us/dotnet/api/system.componentmodel.inotifypropertychanged?view=net-9.0)
This is a standard pattern on C# for a class to notify of changes on their properties.

Once you have that, you can either write a simple event delegate that assigns the appropriate `BehaviorSubject` when an event triggers, or do some reactive subscribing with [`Observable.FromEventPattern`](https://learn.microsoft.com/es-es/previous-versions/dotnet/reactive-extensions/hh211984(v=vs.103))

When the whole object is changed, then it's easy. That's when the setter of the StimulusSequence property triggers, so you just unsubscribe from the notification of the old object, then replace it with the new, and resubscribe to the new one.

bparks13, 2025-08-27:
Notes from technical meeting:
- Following Aaron’s comment on using INotifyPropertyChanged, see if that is a useful pattern that we can add directly to the HS64 GUI PR (#459)
- In the future, if we can attach a GUI to the StimulusSequence source object, users can use the same interface for creating stimulus sequences and then choose which one to feed into the configure node